ROBOTIS-GIT/turtlebot3_manipulation #5: open_manipulator_with_tb3

dltmdwls25, 2018-10-23:
I'm implementing it with a real manipulator, but I'm not sure how to move the actual manipulator.
Can you tell me how to run it?

실제 매니퓰레이터를 가지고 구현 중에 있는데 실제 매니퓰레이터를 움직이는 방법을 잘 모르겠다.
구동하는 방법 알려주실 수 있나요?

routiful, 2018-10-23:
Hi :)

You can find related wiki in emanual. I attached link below

http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation

dltmdwls25, 2018-10-23:
http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation i know this link  
But there is only simulation data. It is difficult because there is no way to actually implement it.

 

http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation

routiful, 2018-10-23:
The packages are used to simulation and actual platform. 
You are supposed to upload firmware (http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#opencr-setup) and bringup it.
Then you can find /joint_states topic and controller it using MoveIt! (http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#moveit)

routiful, 2018-10-23:
Yes, we only support to that. It is resonable because TB3 has OpenCR and controlled by it.
But if you want to using only laptop, you can make your own packages that includes DynamixelSDK or Dynamixel-Workbench to control Manipulator and TB3.

dltmdwls25, 2018-10-23:
한국에 계시면 전화로 물어봐도 될까요?

dltmdwls25, 2018-10-29:
Currently I am trying to control openmanipulator with tb3 with opencr, but turtlebot3_with_open_manipulator_core seems to have a problem.
I downloaded the version of develop and tried to use it, but the same error appears. 
i tried it on another computer
https://github.com/ROBOTIS-GIT/OpenCR/commit/dc825997581e0f845e9d9b17edbfd5d4e9ee3e82

routiful, 2018-10-30:
Because library file is modified, you are supposed to change library folder in Arduino.
I attached a reference how to setup

https://community.robotsource.org/t/howto-adding-multiple-opencr-boards-to-the-arduino-ide/1497

closed 2018-12-12 by gpwhd07