ROBOTIS-GIT/turtlebot3_manipulation #6: open_manipulator_msgs/State.h

Fred159, 2019-01-07:
#include <open_manipulator_msgs/State.h>

`/home/ming/catkin_ws/src/open_manipulator_with_tb3/open_manipulator_with_tb3_tools/src/pick.cpp:27:41: fatal error: open_manipulator_msgs/State.h: No such file or directory
compilation terminated.
open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/pick.dir/build.make:62: recipe for target 'open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/pick.dir/src/pick.cpp.o' failed
make[2]: *** [open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/pick.dir/src/pick.cpp.o] Error 1
CMakeFiles/Makefile2:5151: recipe for target 'open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/pick.dir/all' failed
make[1]: *** [open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/pick.dir/all] Error `

I want to know, there is no 'State.h' in 'open_manipulator_msgs/State.h'. Then how to call that file?

routiful, 2019-01-14:
Hi, I updated OpenManipulator with TB3 into master branch.
But open_manipulator_perception is ready for release. If you can't wait, please download that develop branch.
Please check e-manual and if you have any questions please re-open issue.
http://emanual.robotis.com/docs/en/platform/openmanipulator/#mobile-manipulation

gpwhd07, 2019-01-29:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-01-29 by gpwhd07

hqf2591102423, 2022-01-20:
i still meet the problem that " fatal error: open_manipulator_msgs/State.h no such file ?
how to solve it

hqf2591102423, 2022-01-20:
i still meet the problem that fatal error: open_manipulator_msgs/State.h: no such file .
how can i solve it?