ROBOTIS-GIT/turtlebot3_manipulation #11: [time out ] the launched topic cannot run normally

liusanchuan, 2019-03-24:
```shell
pi@raspberrypi:~ $ ROS_NAMESPACE=om_with_tb3 roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:=om_with_tb3 set_lidar_frame_id:=om_with_tb3/base_scan
... logging to /home/pi/.ros/log/b8840526-4b70-11e9-9f4d-b88198e513d9/roslaunch-raspberrypi-1864.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.104:41515/

SUMMARY
========

PARAMETERS
 * /om_with_tb3/turtlebot3_core/baud: 115200
 * /om_with_tb3/turtlebot3_core/port: /dev/ttyACM0
 * /om_with_tb3/turtlebot3_core/tf_prefix: om_with_tb3
 * /om_with_tb3/turtlebot3_lds/frame_id: om_with_tb3/base_...
 * /om_with_tb3/turtlebot3_lds/port: /dev/ttyUSB0
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /om_with_tb3/
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.0.103:11311

process[om_with_tb3/turtlebot3_core-1]: started with pid [1873]
process[om_with_tb3/turtlebot3_lds-2]: started with pid [1874]
process[om_with_tb3/turtlebot3_diagnostics-3]: started with pid [1875]
[INFO] [1553129085.997015]: ROS Serial Python Node
[INFO] [1553129086.437193]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1553129088.882230]: Note: publish buffer size is 1024 bytes
[INFO] [1553129088.885868]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1553129089.074724]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1553129089.283278]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1553129089.879887]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1553129090.151912]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1553129091.941443]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1553129092.081575]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1553129092.157014]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1553129095.618726]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1553129098.160073]: Note: subscribe buffer size is 1024 bytes
[INFO] [1553129098.164505]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1553129098.543246]: Setup subscriber on joint_trajectory_point [std_msgs/Float64MultiArray]
[INFO] [1553129099.496130]: Setup subscriber on joint_move_time [std_msgs/Float64]
[INFO] [1553129099.606992]: Setup subscriber on gripper_position [std_msgs/Float64MultiArray]
[INFO] [1553129099.724694]: Setup subscriber on gripper_move_time [std_msgs/Float64]
[INFO] [1553129099.804566]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1553129099.892196]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1553129099.964760]: Setup subscriber on reset [std_msgs/Empty]
[WARN] [1553129100.043402]: Failed to get param: timeout expired
[INFO] [1553129100.049559]: Setup TF on Odometry [odom]
[INFO] [1553129100.055617]: Setup TF on IMU [imu_link]
[INFO] [1553129100.061629]: Setup TF on MagneticField [mag_link]
[INFO] [1553129100.067506]: Setup TF on JointState [base_link]
[INFO] [1553129100.082390]: --------------------------
[INFO] [1553129100.088103]: Connected to OpenCR board!
[INFO] [1553129100.094154]: This core(v2.0.0) is compatible with TB3 with OpenManipulator
[INFO] [1553129100.100096]: --------------------------
[INFO] [1553129100.104889]: Start Calibration of Gyro
[INFO] [1553129100.108681]: Calibration End

```

there is  `[WARN] [1553129100.043402]: Failed to get param: timeout expired`
which lead to `[INFO] [1553129100.049559]: Setup TF on Odometry [odom]
[INFO] [1553129100.055617]: Setup TF on IMU [imu_link]` 
so the node don't have the namespace。
But if I Reset OpenCR some times ,this warning will miss.
I have sychronic the time use ` sudo ntpdate ntp.ubuntu.com` , still always appear this warning .

routiful, 2019-03-25:
We have been debugged that issues. It may depends communication delay between OpenCR and main PC.
You can find many solution in answer.ros.org. But I couldn't see correct answers because of unexpected development environments.
I suggest that you can add more delay on `waitForSerialLink` function

https://github.com/ROBOTIS-GIT/OpenCR/blob/2bf9bc3a88d50a335797e6cde655f18ecd1dc55f/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_with_open_manipulator/turtlebot3_with_open_manipulator_core/turtlebot3_with_open_manipulator_core.ino#L781

liusanchuan, 2019-03-26:
ok thank you, I'll have a try .

liusanchuan, 2019-03-28:
I have increase the wait serial‘s delay time to 30 ，and which condition“time out” still   exist。

```c
void waitForSerialLink(bool isConnected)
{
  static bool wait_flag = false;
  
  if (isConnected)
  {
    if (wait_flag == false)
    {      
      delay(30);# add more time 

      wait_flag = true;
    }
  }
  else
  {
    wait_flag = false;
  }
}
```

liusanchuan, 2019-04-19:
Is there any resolution for this Issue, or It's my individual problem？I‘m confused

rjshim, 2019-10-10:
@liusanchuan 

Would you let us know if you have figured out a solution? 

Regards,
Ryan

rjshim, 2019-10-18:
As there have been no updates, I am closing this issue.

closed 2019-10-18 by rjshim