ROBOTIS-GIT/turtlebot3_manipulation #12: [bug report] I found some bug,when i use this reposity

liusanchuan, 2019-03-28:
###  map.yaml
[the absolute link need change ](image: /home/darby/maps/turtlebot3_rooms/map.pgm # Need to change path when you use it)

gpwhd07, 2019-04-08:
Hi @liusanchuan 
Sorry for late. 
If you want to use this package, you have to make a map file, and need to change path when you use it.
About Map file setting, you can refer [e-manual for OpenManipulator with TB3](http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation) and [e-manual for tb3](http://emanual.robotis.com/docs/en/platform/turtlebot3/basic_operation/#basic-operation).

RyanJaehyunShim, 2019-04-11:
@liusanchuan 

Change it as below, then it will work.

/home/computer_name/catkin_ws/src/TB_OM/open_manipulator_with_tb3/open_manipulator_with_tb3_tools/maps/map.pgm

Note that you should modify your computer_name properly (you can easily check it by typing pwd in Terminal. 

Regards,
Ryan

RyanJaehyunShim, 2019-04-11:
@liusanchuan

Change it as below, then it will work.

/home/($PATH_TO_open_manipulator_with_tb3)/open_manipulator_with_tb3_tools/maps/map.pgm

Note that you should modify ($PATH_TO_open_manipulator_with_tb3) properly.

Regards,
Ryan

RyanJaehyunShim, 2019-05-16:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-16 by RyanJaehyunShim