ROBOTIS-GIT/turtlebot3_manipulation #13: control gripper from Python

BeysensJona, 2019-04-11:
I followed this tutorial to move the manipulator when it is placed on the Turtlebot 3: https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html, when I load this launch file on the Turtlebot: 
ROS_NAMESPACE=om_with_tb3 roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:=om_with_tb3 set_lidar_frame_id:=om_with_tb3/base_scan
and this launch file on the remote PC: 
roslaunch open_manipulator_with_tb3_tools manipulation.launch use_platform:=true . Moving the joints of the arm itself works with this code. But how can we move the gripper from Python? 
It seems that there is no group for the gripper, but only for the arm so we cannot use the moveit_commander.MoveGroupCommander(group_name).

RyanJaehyunShim, 2019-04-12:
@BeysensJona 

Unfortunately, we only support the tutorials introduced on our e-manual. However, I believe you can refer to the link below and find some clues for your own work. 
(https://ros-planning.github.io/moveit_tutorials/doc/move_group_interface/move_group_interface_tutorial.html)

Best regards,
Ryan

RyanJaehyunShim, 2019-04-12:
Additionally, you will be able to get some clues from our e-manual. Please refer to the explanation about gripper control in the link below.
(http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#moveit)

Regards,
Ryan

BeysensJona, 2019-04-12:
Hi Ryan, 

Thanks for your reply. However, when following the e-manual I can also not move or get information about the gripper. After initializing all the launch files, when I execute following command: 
[image]
I get the following error: ERROR: service [/om_with_tb3/gripper] responded with an error: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0). 
Furthermore, the rosservice call command seems to acquire info about the gripper, but how to set the how to open and close the gripper?

BeysensJona, 2019-04-16:
Do you have a clue why this exception is thrown? I don't know where this exception is coming from. I followed all steps in the tutorial.

RyanJaehyunShim, 2019-04-17:
@BeysensJona 

Please bring up a turtlebot+openmanipulator model on Gazebo and try below.
$ rostopic pub /om_with_tb3/gripper_position/command std_msgs/Float64 0.1

Regards,
Ryan

RyanJaehyunShim, 2019-05-16:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if you have any other questions. Thanks.

closed 2019-05-16 by RyanJaehyunShim