ROBOTIS-GIT/turtlebot3_manipulation #14: launch error

liusanchuan, 2019-05-05:
while i launch ` ROS_NAMESPACE=om_with_tb3 roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:=om_with_tb3 set_lidar_frame_id:=om_with_tb3/base_scan` 
there is a warn occur, and i want know what the firmware and the software represent?
```
[ WARN] [1557021239.746816542]: This firmware(v2.0.0) isn't compatible with this software (v1.1.0)
[ WARN] [1557021239.747097329]: You can find how to update its in `FAQ` section(turtlebot3.robotis.com)
```

RyanJaehyunShim, 2019-05-07:
@liusanchuan 

As the warning says, refer to the following link and update the firmware. 
(http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#faq)

Regards,
Ryan

liusanchuan, 2019-05-09:
i did as the FAQ say ,but problem still exit 
[ WARN] [1557021239.746816542]: This firmware(v2.0.0) isn't compatible with this software (v1.1.0)

RyanJaehyunShim, 2019-05-16:
@liusanchuan 

Update the software and see if it works. Check out the link below.
(http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#how-to-update-software)

Regards,
Ryan

rjshim, 2019-10-10:
As there have been no updates on this issue, I am closing this issue.

Ryan

closed 2019-10-10 by rjshim