ROBOTIS-GIT/turtlebot3_manipulation #19: This firmware(v2.0.1) isn't compatible with this software (v1.1.0)

rgiordani71, 2019-11-30:
Hi updated the SW and the firmware using the shell scripts provided.
Here the logs

pi@turtlebot:~ $ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2 && tar -xvf opencr_update.tar.bz2 && cd ./opencr_update && ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr && cd ..
--2019-11-27 02:55:46--  https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2
Resolving github.com (github.com)... 140.82.118.3
Connecting to github.com (github.com)|140.82.118.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR-Binaries/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2 [following]
--2019-11-27 02:55:47--  https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR-Binaries/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.240.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.240.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 462386 (452K) [application/octet-stream]
Saving to: ‘opencr_update.tar.bz2’

opencr_update.tar.bz2     100%[==================================>] 451.55K   627KB/s    in 0.7s    

2019-11-27 02:55:48 (627 KB/s) - ‘opencr_update.tar.bz2’ saved [462386/462386]

opencr_update/burger_turtlebot3_core.ino.bin
opencr_update/om_with_tb3.opencr
opencr_update/waffle_turtlebot3_core.ino.bin
opencr_update/opencr_ld_shell_arm
opencr_update/update.sh
opencr_update/
opencr_update/released_1.2.5.txt
opencr_update/waffle.opencr
opencr_update/opencr_ld_shell_x86
opencr_update/burger.opencr
opencr_update/open_manipulator_turtlebot3_core.ino.bin
armv7l
arm
OpenCR Update Start..
opencr_ld_shell ver 1.0.0
opencr_ld_main 
[  ] file name   	: om_with_tb3.opencr 
[  ] file size   	: 206 KB
[  ] fw_name     	: om_with_tb3 
[  ] fw_ver      	: 2.0.1 
[OK] Open port   	: /dev/ttyACM0
[  ]
[  ] Board Name  	: OpenCR R1.0
[  ] Board Ver   	: 0x17020800
[  ] Board Rev   	: 0x00000000
[OK] flash_erase 	: 0.96s
[OK] flash_write 	: 2.21s 
[OK] CRC Check   	: 14127B2 14127B2 , 0.008000 sec
[OK] Download 
[OK] jump_to_fw 
---------------------------------------------------------------

and this is the logs error during the bringup
SUMMARY
========

PARAMETERS
 * /om_with_tb3/turtlebot3_core/baud: 115200
 * /om_with_tb3/turtlebot3_core/port: /dev/ttyACM0
 * /om_with_tb3/turtlebot3_core/tf_prefix: om_with_tb3
 * /om_with_tb3/turtlebot3_lds/frame_id: om_with_tb3/base_...
 * /om_with_tb3/turtlebot3_lds/port: /dev/ttyUSB0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /om_with_tb3/
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.178.83:11311

process[om_with_tb3/turtlebot3_core-1]: started with pid [2393]
process[om_with_tb3/turtlebot3_lds-2]: started with pid [2394]
process[om_with_tb3/turtlebot3_diagnostics-3]: started with pid [2395]
[INFO] [1574853822.489446]: ROS Serial Python Node
[INFO] [1574853822.567683]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1574853824.731906]: Note: publish buffer size is 1024 bytes
[INFO] [1574853824.734646]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1574853824.768773]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1574853824.864398]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1574853824.893202]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1574853824.930066]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1574853824.946887]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1574853824.961481]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1574853824.978443]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1574853827.293922]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1574853827.325325]: Note: subscribe buffer size is 1024 bytes
[INFO] [1574853827.327018]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1574853827.358766]: Setup subscriber on joint_trajectory_point [std_msgs/Float64MultiArray]
[INFO] [1574853827.390090]: Setup subscriber on joint_move_time [std_msgs/Float64]
[INFO] [1574853827.433733]: Setup subscriber on gripper_position [std_msgs/Float64MultiArray]
[INFO] [1574853827.481831]: Setup subscriber on gripper_move_time [std_msgs/Float64]
[INFO] [1574853827.513743]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1574853827.545626]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1574853827.601364]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1574853827.845636]: Setup TF on Odometry [om_with_tb3/odom]
[INFO] [1574853827.851397]: Setup TF on IMU [om_with_tb3/imu_link]
[INFO] [1574853827.857301]: Setup TF on MagneticField [om_with_tb3/mag_link]
[INFO] [1574853827.863245]: Setup TF on JointState [om_with_tb3/base_link]
[INFO] [1574853827.875962]: --------------------------
[INFO] [1574853827.881860]: Connected to OpenCR board!
[INFO] [1574853827.887690]: This core(v2.0.1) is compatible with TB3 with OpenManipulator
[INFO] [1574853827.893391]: --------------------------
[INFO] [1574853827.896981]: Start Calibration of Gyro
[INFO] [1574853830.406885]: Calibration End
[ WARN] [1574853830.988542702]: This firmware(v2.0.1) isn't compatible with this software (v1.1.0)
[ WARN] [1574853830.988814629]: You can find how to update its in `FAQ` section(turtlebot3.robotis.com)

So it seams that the software update procedure doesn't update the current installation on ubuntu 16
http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#how-to-update-software

Let me know how to proceed.
Thanks

gerenb, 2019-12-01:
I have the same problem.

rgiordani71, 2019-12-02:
any update, comment?

The3Designer, 2020-01-15:
Same issue here on a Turtlebot3 Waffle Pi with OpenMANIPULATOR-X. Any updates on a possible fix/solution?

I also think that because of this issue, I'm not able to control to joints of the arm. It's possible to read their current status, but adjusting the angles of each Dynamixel is impossible.

rjshim, 2020-02-14:
Duplicate of https://github.com/ROBOTIS-GIT/open_manipulator_with_tb3/issues/21#issuecomment-586067820

closed 2020-02-14 by rjshim