ROBOTIS-GIT/turtlebot3_manipulation #21: Software and MoveIt problem on Open Manipulator with Turtlebot 3 Waffle Pi

krizchong, 2020-01-02:
I did follow emanual in Manipulation of Turtlebot3 to setup Open Manipulator with Turtlebot3 Waffle Pi.
But I face problems when follow emanual to run.

On Remote PC:
1st Terminal, run "roscore", no problem.
2nd Terminal, ssh to TB3, run "$ ROS_NAMESPACE=om_with_tb3 roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:=om_with_tb3 set_lidar_frame_id:=om_with_tb3/base_scan", got warning:
[image]

I read forum and github, this problem has been raised for a while, any solution or update?

3rd Terminal, run "ROS_NAMESPACE=om_with_tb3 roslaunch open_manipulator_with_tb3_tools om_with_tb3_robot.launch", no problem.
4th Terminal, run "roslaunch open_manipulator_with_tb3_tools manipulation.launch use_platform:=true", got warning:
[image]

and in the RVIZ, we cannot move the Manipulator on interface as shown in the emanual
[image]

Which setup goes wrong and causes these warning and cannot MoveIt? 
( I did try the rosservice to move joints, no problem )

krizchong, 2020-01-07:
Anyone facing the same problem or any suggestion to investigate?

xEvheMary, 2020-01-23:
I'm having the same problem.
I can move only the turtlebot using the turtlebot teleop key which mean that the bringup shouldn't be the issue.
I tried using the UI, but the joints value is stuck at current condition and cannot be changed.
Also asking for solutions here if someone figured it out already.

100kimch, 2020-02-04:
Same problem too.

The3Designer, 2020-02-07:
Seems to be the same issue as this one: https://github.com/ROBOTIS-GIT/open_manipulator_with_tb3/issues/19

Perhaps @rjshim has an update on this issue?

rjshim, 2020-02-14:
@krizchong @xEvheMary @100kimch @The3Designer 

Can you guys update the open_manipulator_with_tb3 and open_manipulator_with_tb3_simulations repositories and test below? 

$ [RemotePC] roscore
$ [TB3] roslaunch turtlebot3_bringup turtlebot3_robot.launch
roslaunch turtlebot3_manipulation_bringup turtlebot3_manipulation_bringup.launch
$ [RemotePC] roslaunch turtlebot3_manipulation_moveit_config move_group.launch
$ [RemotePC] roslaunch turtlebot3_manipulation_moveit_config moveit_rviz.launch

You don't have to care about the warning code from the Turtlebot. Let me know if it still doesn't work.

Regards,
Ryan

The3Designer, 2020-02-14:
@rjshim  I'm currently not in the position to access the Turtlebot, but I'll update it in about a week. Then I'll let you know if it works. Thanks for the effort so far!

rjshim, 2020-02-20:
As there has been no update, I will close this issue. 
Ryan.

babakc, 2020-02-21:
FYI this worked for me. It's worth noting that the emanual needs to be updated.

xEvheMary, 2020-02-21:
can i ask the difference between the turtlebot3_manipulation and open_manipulator_with_tb3?
should i add both?

rjshim, 2020-02-24:
@xEvheMary @babakc 

We changed the package names to turtlebot3_manipulation. The instruction written on the eManual website will be updated soon. If you would like to test your TB3 + OMX robot, follow the following [link](https://github.com/ROBOTIS-GIT/turtlebot3_manipulation/issues/21#issuecomment-586067820).

Ryan

The3Designer, 2020-02-29:
@rjshim Sorry for the delay, but I can confirm that the arm can now indeed be controlled. Thanks for the fix(es)!

rjshim, 2020-03-01:
@The3Designer Great :) let us know if you have any further questions. 👍

babakc, 2020-03-05:
I'm struggling to build SLAM maps, navigate, and run the pick and place sim without the updated doco. When will this be updated?

babakc, 2020-03-12:
@rjshim , this should be reopened until the documentation is updated and tested. I can move the arm but navigation and the gripper doesn't work.

chris19971, 2020-03-24:
@rjshim Currently I am working on my bachelor thesis and have the same problem as @babakc !
I can move the arm with the script above but I also need to control the gripper for picking up things.

xEvheMary, 2020-03-25:
Currently i'm also having the similar problem, which is the gripper. It can be closed from open position, but then it cannot be opened again. I also want to ask whether anyone tried controlling the manipulator using it's keyboard teleop node?

ROBOTIS-Will, 2020-03-26:
@babakc 
SLAM and Navigation with OpenMANIPULATOR-X instructions will be updated soon. For the pick and place, there will be a major source code change. As an updated pick and place example, we will soon release the detailed manual regarding [Turtlebot3 Home Service Challenge](https://github.com/ROBOTIS-GIT/turtlebot3_home_service_challenge)
@xEvheMary 
Are you using this package for TurtleBot3 + OpenMANIPULATOR-X? or is it solely a seperate OpenMANIPULATOR-X?

babakc, 2020-03-26:
Sorry, i should have updated, i finally got this working. Still a work in progress but... [TB3 with OM video](https://photos.app.goo.gl/6RixUBgnN1vMcew77)

xEvheMary, 2020-03-26:
@ROBOTIS-Will
Thx for answering. I controlled the gripper using the UI from the new package so i thought it should be fine but it can only open. The same happened when i used the OM's UI. 
Since the new UI looks the same like openmanipulator's UI, i assume it can also be controlled by OM's teleop and also use the same topic and messages, but i haven't tried it yet.

xEvheMary, 2020-04-02:
Update:
Unable to use OpenManipulator's teleop, probably because of the difference in service/topic used (?)
Will each launch file/node explained in the documentation? I'm not proficient in cpp and having a hard time making a similar program using python

ROBOTIS-Will, 2020-04-08:
@xEvheMary 
The previous version of teleop for OM is not applicable for TurtleBot3 manipulation package as the new package is running based on move_group.
Thank you.

krizchong, 2020-05-06:
@ROBOTIS-Will 
Hi, Will. I tried the new Turtlebot Manipulation packages on the ROBOTIS GUI Controller.
In the first few try there is no problem to control the manipulator with GUI but suddenly error pop out

[ INFO] [1588754769.188051335]: Didn't received robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1588754769.188237054]: Failed to fetch current robot state
[ INFO] [1588754770.188450484]: Didn't received robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1588754770.188556746]: Failed to fetch current robot state
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[turtlebot3_manipulation_gui-1] process has died [pid 15007, exit code -6, cmd /home/robotclub/catkin_ws/devel/lib/turtlebot3_manipulation_gui/turtlebot3_manipulation_gui __name:=turtlebot3_manipulation_gui __log:=/home/robotclub/.ros/log/3ba3ccae-8f70-11ea-b336-0cd292ac365e/turtlebot3_manipulation_gui-1.log].
log file: /home/robotclub/.ros/log/3ba3ccae-8f70-11ea-b336-0cd292ac365e/turtlebot3_manipulation_gui-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

I tried many time even with "sudo ntpdate ntp.ubuntu.com" on both PC and TB3 also always come out these error messages after running the GUI for a while. Any idea what is the problem.

I just followed the instruction in the emanual...

ROBOTIS-Will, 2020-05-08:
@krizchong 
We'll check this issue and get back to you. Thank you.

ashekim, 2020-05-12:
@krizchong 
Thank you for your questions. 

Please check your turtlebot3 Wi-Fi connection. If Wi-Fi connection is lost, the above problem may occur. 
Let me know if above still doesn't work.

Regards,
Ashe

The3Designer, 2020-05-31:
> 
> 
> Sorry, i should have updated, i finally got this working. Still a work in progress but... [TB3 with OM video](https://photos.app.goo.gl/6RixUBgnN1vMcew77)

I don't want to hijack this issue tracker, but did you get the Home Service Challenge working by just following the tutorial (http://emanual.robotis.com/docs/en/platform/turtlebot3/home_service_challenge/)? Did it work at once, because I'm having an issue where it is not consistently working (issue tracker: https://github.com/ROBOTIS-GIT/turtlebot3_home_service_challenge/issues/9). So I'm just wondering what you did to make it work?

ROBOTIS-Will, 2020-06-03:
In many cases, network and time sync are causing unstable operation. If possible, try running the example with ethernet cables connected to both TurtleBot3 and the Remote PC to see if this helps to resolve the issue.
Thank you.

The3Designer, 2020-06-04:
@ROBOTIS-Will, my problem is solved. It was an issue where the objects in the room where not exactly the same as when the SLAM map was generated. Therefore, the costmap had troubles being generated. Thanks for all the suggestions!

xEvheMary, 2020-06-09:
I have a problem with the gripper of the OpenMnaipulator
it won't move and the joint value always show -0.032
i used python moveit_commander to move and check the joint value and it deliver the same values
Is it a hardware or software problem?

ROBOTIS-Will, 2020-06-10:
@The3Designer 
Glad to hear that you were able to figure out the problem.

@xEvheMary 
Please note that OpenMANIPULATOR in home service challenge use different group names for manipulator(`arm`) and the gripper(`gripper`). If you've used the group name `arm` for the gripper, it won't move.

xEvheMary, 2020-06-10:
@ROBOTIS-Will 
i used the group named gripper
in my python code, it return -0.032 for joint gripper and gripper sub
the turtlebot3_manipulation_gui also return the same values
i tried changing the values to 0 using moveit_commander
the planner return a success, but the gripper won't move

EndlessLoops, 2020-07-17:
> @ROBOTIS-Will
> Hi, Will. I tried the new Turtlebot Manipulation packages on the ROBOTIS GUI Controller.
> In the first few try there is no problem to control the manipulator with GUI but suddenly error [...]

I got the same problem.Did you found the reason of the error "terminate called after throwing an instance of 'std::out_of_range'"
@ROBOTIS-Will 
```
lalala@lalala-Inspiron-3576:~$ roslaunch turtlebot3_home_service_challenge_manager manager.launch                
... logging to /home/lalala/.ros/log/9389af24-c71a-11ea-a4d9-00044b8d7728/roslaunch-lalala-Inspiron-3576-7069.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://192.168.0.106:38941/
SUMMARY
========
PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /tb3_hsc/tb3_hsc_manager_node/task_data_path: /home/lalala/catk...
NODES
  /tb3_hsc/
    tb3_hsc_manager_node (turtlebot3_home_service_challenge_manager/tb3_hsc_manager_node)
ROS_MASTER_URI=http://192.168.0.94:11311
process[tb3_hsc/tb3_hsc_manager_node-1]: started with pid [7078]
[ INFO] [1594873641.044579989]: Loading robot model 'turtlebot3_manipulation'...
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 478) >= this->size() (which is 360)
[tb3_hsc/tb3_hsc_manager_node-1] process has died [pid 7078, exit code -6, cmd /home/lalala/catkin_ws/devel/lib/turtlebot3_home_service_challenge_manager/tb3_hsc_manager_node __name:=tb3_hsc_manager_node __log:=/home/lalala/.ros/log/9389af24-c71a-11ea-a4d9-00044b8d7728/tb3_hsc-tb3_hsc_manager_node-1.log].
log file: /home/lalala/.ros/log/9389af24-c71a-11ea-a4d9-00044b8d7728/tb3_hsc-tb3_hsc_manager_node-1*.log
[tb3_hsc/tb3_hsc_manager_node-1] restarting process
process[tb3_hsc/tb3_hsc_manager_node-1]: started with pid [7091]
[ INFO] [1594873648.089901616]: Loading robot model 'turtlebot3_manipulation'...
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 478) >= this->size() (which is 360)
[tb3_hsc/tb3_hsc_manager_node-1] process has died [pid 7091, exit code -6, cmd /home/lalala/catkin_ws/devel/lib/turtlebot3_home_service_challenge_manager/tb3_hsc_manager_node __name:=tb3_hsc_manager_node __log:=/home/lalala/.ros/log/9389af24-c71a-11ea-a4d9-00044b8d7728/tb3_hsc-tb3_hsc_manager_node-1.log].
log file: /home/lalala/.ros/log/9389af24-c71a-11ea-a4d9-00044b8d7728/tb3_hsc-tb3_hsc_manager_node-1*.log
^C[tb3_hsc/tb3_hsc_manager_node-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

EndlessLoops, 2020-07-21:
> I got the same problem.Did you found the reason of the error "terminate called after throwing an instance of 'std::out_of_range'"
> @ROBOTIS-Will
>
> ```
> lalala@lalala-Inspiron-3576:~$ roslaunch [...]

I found the reason why I got the error.It was because I have modified the source code of hls lidar .When I replaced it with the original code,the program worked fine.

ROBOTIS-Will, 2020-07-24:
@EndlessLoops I'm sorry about delayed reponse, and thanks for the update. Glad that it has been resolved.

JaehyunShim, 2020-10-16:
Since there haven't been any updates, I will close this issue. If you have any other questions, please, open a new issue thread

closed 2020-10-16 by JaehyunShim