ROBOTIS-GIT/turtlebot3_manipulation #31: om_with_tb3_remote.launch is missing

skRoh, 2020-03-09:
**eManual - Turtlebot3**
12.[ROS1]Manipulation
12.5 bringup
  [Remote PC] Launch Launch robot_state_publisher node
------------------------------------------------------------------------------------------------------------------------------------
  $ _ROS_NAMESPACE=om_with_tb3 roslaunch open_manipulator_with_tb3_tools om_with_tb3_remote.launch_
-->
  RLException: [om_with_tb3_remote.launch] is neither a launch file in package   [open_manipulator_with_tb3_tools] nor is [open_manipulator_with_tb3_tools] a launch file name

_~/catkin_ws/src/open-manix/src/open_manipulator_with_tb3/open_manipulator_with_tb3_tools/launch$ls
amcl.launch          navigation.launch  slam.launch
manipulation.launch  robot.launch       task_controller.launch
moveit_rviz.launch   rooms.launch_
------------------------------------------------------------------------------------------------------------------------------------
there is no om_with_tb3_remote.launch file in https://github.com/ROBOTIS-GIT/turtlebot3_manipulation/tree/melodic-devel

ROBOTIS-Will, 2020-04-08:
@skRoh 
Hi,
`om_with_tb3` package has been revised and renamed to `turtlebot3_manipulation`.
Since teleop is 
Please use teleop from the turtlebot3_teleop package in turtlebot3 metapackage.
```
$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
```
Thank you.

ROBOTIS-Will, 2020-05-20:
This thread will be closed as there wasn't any activity recently. Please feel free to reopen when needed.

closed 2020-05-20 by ROBOTIS-Will