ROBOTIS-GIT/turtlebot3_manipulation #37: Unsure of which service messages to use to move joints and gripper using rospy

nicholashojunhui, 2020-04-29:
We would like to service messages to move the joints and gripper of the OM using rospy under this new package. In the older package version (i.e. open_manipulator_with_tb3), we manage to do this with '/arm/moveit/set_joint_position' and '/om_with_tb3/gripper'. However, for this newer package, we are clueless which service messages to use to perform these tasks. we have check out the available service messages while running the Gazebo and the move_group nodes, and we cannot find anything that is relevant to move the joints and gripper of the OM under the move_group section (below).
[image]

Please kindly assist and guide us in this matter. Thank you and much appreciate your time.

On a side note, the RViz and GUI functions are working well for us; just unsure which service messages to use to move the arm under this new package. And we think the command given in your website to run the Gazebo for TB3 with OM is not updated; it should be:
$ roslaunch turtlebot3_manipulation_gazebo turtlebot3_manipulation_gazebo.launch

instead of
$ roslaunch turtlebot3_manipulation_gazebo turtlebot3_manipulation.launch

This launch file also has some minor syntax error that you might want to rectify as of the time we updated and installed this package (i.e. a few days ago).

ROBOTIS-Will, 2020-05-12:
@nicholashojunhui 
Thank you for reporting the wrong command in the eManual.
The recent changes in turtlebot3_manipulation package, moveit planning interface is adopted for more general purpose.
Unlike the previous code where we used messages, you should inherit [moveit interface classes](http://docs.ros.org/jade/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroup.html).
Please refer to below section where gripper control is implemented in the GUI program.
https://github.com/ROBOTIS-GIT/turtlebot3_manipulation/blob/master/turtlebot3_manipulation_gui/src/qnode.cpp#L200

nicholashojunhui, 2020-05-12:
@ROBOTIS-Will 
Thank you for your response. Noted, I will try it. Much appreciated your time

nicholashojunhui, 2020-05-14:
@ROBOTIS-Will 
I have tried the method that you have mentioned on Gazebo as well on the Physical TB3_with_OM, and it works well. Took reference from this website: http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html

Thank you and much appreciated your time in this matter.

closed 2020-05-14 by nicholashojunhui