ROBOTIS-GIT/turtlebot3_manipulation #44: Manipulation of Turtlebot3 with Open Manipulator. joint state error

jellegroothuis, 2022-06-01:
I did follow emanual in Manipulation of Turtlebot3 to setup Open Manipulator with Turtlebot3 Waffle.
But I face problems when follow emanual to run.

On Remote PC:
roslaunch turtlebot3_bringup turtlebot3_robot.launch, No problem
roslaunch turtlebot3_manipulation_bringup turtlebot3_manipulation_bringup.launch, No problem
roslaunch turtlebot3_manipulation_moveit_config move_group.launch, I got a warning:
[image]

does anyone know what i'm doing wrong? or how can I fix this problem? my knowledge of ROS is currently not good enough to remedy this.

ROBOTIS-Will, 2022-06-02:
Hi @jellegroothuis 
Which ROS do you use?
You should run the bringup command on your Turtlebot3 SBC(Raspberry Pi), not on your Remote PC
```bash
$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
```

jellegroothuis, 2022-06-03:
the problem is solved. the wrong software was on the openCR.

turtlebot3 with openmanipulator was needed while turtlebot 3 core was in the manual.

goekce, 2023-08-14:
@JaehyunShim please close the issue if your problem is solved.

closed 2024-06-13 by sunghowoo