ROBOTIS-GIT/turtlebot3_manipulation #51: Failure with ros2 launch turtlebot3_manipulation_bringup hardware.launch.py

UniBot50, 2022-12-09:
I'm working with turtlebot 3 waffle and the OpenMANIPULATOR-X mounted in top. I installed everything as described in the e-manual. But when I want to launch the bring up with <ros2 launch turtlebot3_manipulation_bringup hardware.launch.py> I get the following error message:  

[image]
  

Somebody what I can do about that?

JBVAkshaya, 2022-12-12:
I am running into the same issue. Were you able to fix this issue? There is a high possibility that as of now this repository is not compatible with ROS 2. From the wiki, the latest compatibility seems to be with ROS Melodic + Ubuntu Bionic.

wecacuee, 2023-01-04:
[hardware.launch.py](https://github.com/ROBOTIS-GIT/turtlebot3_manipulation/blob/foxy-devel/turtlebot3_manipulation_bringup/launch/hardware.launch.py) includes [base.launch.py](https://github.com/ROBOTIS-GIT/turtlebot3_manipulation/blob/foxy-devel/turtlebot3_manipulation_bringup/launch/base.launch.py) which needs `FindExecutable(name='xacro')`. 

Make sure ros-foxy-xacro is installed
```
sudo apt install ros-foxy-xacro
```

closed 2023-02-16 by UniBot50