ROBOTIS-GIT/turtlebot3_manipulation #57: Jerked Arm movements  vs. roadmap

lmendyk, 2023-02-22:
I managed to start the system based on a branch applying the ROS2 control and use MoveIt2 for planning the arm  movements but the result is very unsmooth movements (the whole tb3_om jerks when moves) .  I don’t know if this is due to some bugs/issues similar to (https://github.com/ROBOTIS-GIT/turtlebot3_manipulation/issues/55#issuecomment-1438289847)   or it is “by design” due to current implementation scope and the improvement is on the roadmap.  In other words is there a roadmap for this project and description of assumption/limitation assumed. 

I’m new to robotics, but my understanding is that to have arm movement smooth, the joint_trajectory_controller may need to use not just “position” but also “velocity”  (and “acceleration”) . Only after that using MoveIt with plugins for trajectory smoothing  can result  with smoot robot movements.  I’m not sure if I correct understand this.

Anyway, is in the scope of this project to enable smoot control of the arm movement using ROS2 controls – or it is out -of scope.

sunghowoo, 2024-09-17:
@lmendyk  Sorry for late response. Could you share a video showing the issue you're experiencing with your robot's movements? It would help us better understand how the movements are not smooth and give us more insight into the problem you're encountering.

sunghowoo, 2025-01-16:
This issue will be closed due to prolonged inactivity. Please try again with the Humble version, and if the issue persists, feel free to reopen it.

closed 2025-01-16 by sunghowoo