ROBOTIS-GIT/turtlebot3_manipulation #59: ROS2 Foxy - MoveIt Simulation "NO PLANNING LIBRARY LOADED" error

sicktronics, 2023-04-14:
Hello,

After following the e-manual installation instructions, I launched a simulation with MoveIt using the following command:

`$ ros2 launch turtlebot3_manipulation_moveit_config moveit_gazebo.launch.py`

The launch is successful, however in the "MotionPlanning" Context panel I get a "NO PLANNING LIBRARY LOADED" error. I tried creating a new from-scratch workspace and encountered the same thing. 

- Distro: ROS2 Foxy
- OS: Ubuntu 20.04

Here is my full terminal output:

```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [38869]
[INFO] [move_group-2]: process started with pid [38871]
[INFO] [robot_state_publisher-3]: process started with pid [38873]
[INFO] [spawner.py-4]: process started with pid [38875]
[INFO] [spawner.py-5]: process started with pid [38877]
[INFO] [spawner.py-6]: process started with pid [38880]
[INFO] [spawner.py-7]: process started with pid [38891]
[INFO] [gzserver-8]: process started with pid [38893]
[INFO] [gzclient   -9]: process started with pid [38896]
[INFO] [spawn_entity.py-10]: process started with pid [38900]
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link base_link had 8 children
[robot_state_publisher-3] Link link1 had 1 children
[robot_state_publisher-3] Link link2 had 1 children
[robot_state_publisher-3] Link link3 had 1 children
[robot_state_publisher-3] Link link4 had 1 children
[robot_state_publisher-3] Link link5 had 3 children
[robot_state_publisher-3] Link end_effector_link had 0 children
[robot_state_publisher-3] Link gripper_left_link had 0 children
[robot_state_publisher-3] Link gripper_right_link had 0 children
[robot_state_publisher-3] Link camera_link had 1 children
[robot_state_publisher-3] Link camera_rgb_frame had 1 children
[robot_state_publisher-3] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher-3] Link caster_back_left_link had 0 children
[robot_state_publisher-3] Link caster_back_right_link had 0 children
[robot_state_publisher-3] Link imu_link had 0 children
[robot_state_publisher-3] Link base_scan had 0 children
[robot_state_publisher-3] Link wheel_left_link had 0 children
[robot_state_publisher-3] Link wheel_right_link had 0 children
[robot_state_publisher-3] [INFO] [1681512808.602366477] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1681512808.605008194] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1681512808.605053101] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1681512808.605071700] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-3] [INFO] [1681512808.605089053] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-3] [INFO] [1681512808.605106092] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher-3] [INFO] [1681512808.605122099] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher-3] [INFO] [1681512808.605138993] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher-3] [INFO] [1681512808.605155144] [robot_state_publisher]: got segment end_effector_link
[robot_state_publisher-3] [INFO] [1681512808.605171450] [robot_state_publisher]: got segment gripper_left_link
[robot_state_publisher-3] [INFO] [1681512808.605188467] [robot_state_publisher]: got segment gripper_right_link
[robot_state_publisher-3] [INFO] [1681512808.605205131] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1681512808.605222064] [robot_state_publisher]: got segment link1
[robot_state_publisher-3] [INFO] [1681512808.605239789] [robot_state_publisher]: got segment link2
[robot_state_publisher-3] [INFO] [1681512808.605257147] [robot_state_publisher]: got segment link3
[robot_state_publisher-3] [INFO] [1681512808.605274065] [robot_state_publisher]: got segment link4
[robot_state_publisher-3] [INFO] [1681512808.605290606] [robot_state_publisher]: got segment link5
[robot_state_publisher-3] [INFO] [1681512808.605330650] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1681512808.605348844] [robot_state_publisher]: got segment wheel_right_link
[move_group-2] [WARN] [1681512808.670592127] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-2] [WARN] [1681512808.671614299] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-2] Parsing robot urdf xml string.
[move_group-2] [INFO] [1681512808.702879983] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-2] [INFO] [1681512808.703893853] [moveit_robot_model.robot_model]: Loading robot model 'turtlebot3_manipulation'...
[spawner.py-7] [INFO] [1681512808.964302724] [spawner_gripper_controller]: Waiting for /controller_manager services
[move_group-2] [WARN] [1681512809.205658927] [moveit_robot_model.robot_model]: Link end_effector_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[... 190 lines cut ...]
[move_group-2] *     - QueryPlannersService
[move_group-2] *     - StateValidationService
[move_group-2] ********************************************************
[move_group-2] 
[move_group-2] [INFO] [1681512840.445038661] [moveit_move_group_capabilities_base.move_group_context]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[move_group-2] [INFO] [1681512840.445066735] [moveit_move_group_capabilities_base.move_group_context]: MoveGroup context initialization complete
[move_group-2] Loading 'move_group/ApplyPlanningSceneService'...
[move_group-2] Loading 'move_group/ClearOctomapService'...
[move_group-2] Loading 'move_group/MoveGroupCartesianPathService'...
[move_group-2] Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
[move_group-2] Loading 'move_group/MoveGroupGetPlanningSceneService'...
[move_group-2] Loading 'move_group/MoveGroupKinematicsService'...
[move_group-2] Loading 'move_group/MoveGroupMoveAction'...
[move_group-2] Loading 'move_group/MoveGroupPlanService'...
[move_group-2] Loading 'move_group/MoveGroupQueryPlannersService'...
[move_group-2] Loading 'move_group/MoveGroupStateValidationService'...
[move_group-2] 
[move_group-2] You can start planning now!
[move_group-2] 

```

One thing that is strange is that this error appears to occasionally self-resolve, but there is no apparent pattern to when this happens (perhaps 1 out of 10 launches). 

Let me know if I can share any other information. Any advice would be appreciated. Thank you!

yeongkyuyoon, 2023-07-27:
@sicktronics Hi, Could you try install below package?

**sudo apt install ros-foxy-ompl**

sunghowoo, 2024-09-17:
I will close this issue due to inactivity. If there are any further problems, please feel free to reopen it

closed 2024-09-17 by sunghowoo