ROBOTIS-GIT/turtlebot3_manipulation #60: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 1.639534 seconds). Stopping trajectory.

nicholashojunhui, 2023-05-10:
Using ROS 1 Noetic Ninjemys, Raspberry Pi 3B+ for turtlebot3 with open manipulator-x

**Commands or instructions that reproduce the issue:**
1. Switched on TB3 (with OM) and activated Bringup
2. Ran Bringup node for OM on TB3 (on new terminal):
$ roslaunch turtlebot3_manipulation_bringup turtlebot3_manipulation_bringup.launch
3. Ran move_group node:
$ roslaunch turtlebot3_manipulation_moveit_config move_group.launch
4.  Ran the below node:
$ rosrun autonomous OM_DEMO.py
OM_DEMO.py code can be found in the following link:
https://github.com/nicholashojunhui/autonomous/blob/main/src/OM_DEMO.py 

**Error messages on terminal:**
"Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was XXXXX seconds). Stopping trajectory."

[image]
[image]
[image]
[image]

**Issues Described in detail:**
1. Unable to move the arm (as part of the Turtlebot3_Manipulation system) properly using "moveit_commander"
2. Keep encountering the following error message: "Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was XXXXX seconds). Stopping trajectory."
3. The 4 normal joints unable to move properly (i.e. jerking manner or will not move) and kept aborting: "ABORTED: TIMED_OUT"
4. Gripper has no issue moving at all unlike the other 4 joints
5. The PC will hang every time after code execution

**Previously used in ROS Kinetic and it works very well**
- No issues at all with the above python node when I used ROS Kinetic

**The weird thing is that I am able to run the above python node in the simulation mode in Gazebo**
- Simulated Turtlebot3_Manipulation in Gazebo has no issues at all with the above python node (using ROS Noetic)

eRobotclub, 2024-01-26:
I am also facing a similar problem when controlling OpenManipulator-X on TB3 with ROS Noetic by GUI interface:

[image]

Any update about this issue?

sunghowoo, 2024-07-18:
@nicholashojunhui Sorry for late reply. There has been a lack of attention to package version management. Switching to the develop branch should resolve this issue:
git clone --branch develop <https://github.com/ROBOTIS-GIT/turtlebot3_manipulation.git>
Demo video: https://drive.google.com/file/d/1WG7eOessCJFXNVrK0Efks_Z46yi0Wjx_/view?usp=sharing

sunghowoo, 2024-09-17:
I am closing this issue due to inactivity.

closed 2024-09-17 by sunghowoo