ROBOTIS-GIT/turtlebot3_manipulation #61: ros2_controller hardware plugin only for open_manipulator_x (ROS2 Humble)

Aki1608, 2023-06-22:
Hello everyone,

This is more like a question than an issue. I have the actual hardware and wanted to know if there is any ros2_controller hardware plugin for the open manipulator arm so that I can control the arm using rviz?

I can move the arm using teleop but I cant control it with Rviz. I think I have to change the hardware plugin in the ros2_controller.xacro file. But I don't have any hardware plugin for open_manipulator_x. In this repo there is one plugin defined in [turtlebot3_manipulation_system.ros2_control.xacro](https://github.com/ROBOTIS-GIT/turtlebot3_manipulation/blob/humble-devel/turtlebot3_manipulation_description/ros2_control/turtlebot3_manipulation_system.ros2_control.xacro) file but it is for turtlebot and open_manipulator arm together. (at line 26, "<plugin>turtlebot3_manipulation_hardware/TurtleBot3ManipulationSystemHardware</plugin>"). Is there any similar plugin for open_manipulator_x arm alone? 

Can anyone help me or guide me in finding/creating the plugin? Or any similar plugin is available?
I am using ROS2 humble on Ubuntu 22.04.

Thank you.

ROBOTIS-Will, 2023-07-20:
Hi @Aki1608 

The related package will soon be available.
Hopefully the TurtleBot3 Manipulation package using MoveIt on RViz for Humble and Iron will be updated by this month.
eManual documentation will need more time though.
Thanks!

Aki1608, 2023-08-04:
Hi @ROBOTIS-Will, Turtlebot3 manipulation package is working for Humble. I can control the arm through moveit2 on RViz.

Is there any update on hardware plugin only for open_manipulator_x? Actually, I want to control only the arm using U2D2 board through RViz, and waffle through openCR.

ROBOTIS-Will, 2023-08-07:
Hi @Aki1608 

Yes, we are currently working with both TurtleBot3 and OpenMANIPULATOR-X update.
Humble is WIP in humble-devel branch of the open_manipulator package
https://github.com/ROBOTIS-GIT/open_manipulator/tree/humble-devel

Aki1608, 2023-08-10:
Okay. Thank you @ROBOTIS-Will.

I have one more question. Is it possible to add virtual joints in the hardware plugin and OpenCR firmware to control the real hardware? 

Actually, I want to control the arm with moveit2 (inverse kinematics). But to my knowledge, the moveit2 has been written such that the arm should have minimum 6 DoF. I tried adding virtual joints in gazebo and I was able to control the arm using moveit2 and Rviz (inverse kinematics). 

I tried adding joints in OpenCR firmware file and also in the hardware plugin but I was not able to successfully control the arm. So before I try any further I wanted to know if it is even possible to add virtual joints and control real hardware?

sunghowoo, 2024-09-17:
It seems your initial question has been resolved, so I will close this issue.

closed 2024-09-17 by sunghowoo