ROBOTIS-GIT/turtlebot3_manipulation #63: Open Manipulator Fails BringUP

gandradeng, 2023-07-01:
Hi everyone, I'm having this errors (exception) when trying to bringup the turtlebot3 with the open manipulator:
$ ros2 launch turtlebot3_manipulation_bringup hardware.launch.py 
  [INFO] [launch]: All log files can be found below /home/dali/.ros/log/2023-07-01-20-56-48-983432-dali-turtlebot-10821
  [INFO] [launch]: Default logging verbosity is set to INFO
  [ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 
  '[<launch.substitutions.text_substitution.TextSubstitution object at 0xffff8f854100>]' not found on the PATH

Data: Turtlebo3 with OpenManipulator X, Humble, Ubuntu 22.04, Raspberry Pi 4.

yeongkyuyoon, 2023-07-10:
Hi, Could you try below command on 'turtlebot3_ws' using Terminal?

$ source install/setup.sh

ozanaki, 2023-07-13:
I have the same issue too. 
@yeongkyuyoon Sourcing the install/setup.bash did not fix the issue.

ROS2 Humble / Ubuntu 22.04 Server / Raspberry Pi 4 / Turtlebot3 Waffle Pi with Manipulator.

`ubuntu@wafflepi:~/turtlebot3_ws$ source install/setup.bash
ubuntu@wafflepi:~/turtlebot3_ws$ ros2 launch turtlebot3_manipulation_bringup hardware.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-07-13-16-58-41-633216-wafflepi-1236
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable '[<launch.substitutions.text_substitution.TextSubstitution object at 0xffffbafb6c50>]' not found on the PATH
ubuntu@wafflepi:~/turtlebot3_ws$`

yeongkyuyoon, 2023-07-17:
@ozanaki 
Perhaps some dependency packages are not installed, resulting in errors when executing the launch file. I identified all the missing packages in the ROS2 Foxy version and conducted tests accordingly.

It is assumed that the Humble version also encounters a similar problem. Please install the following dependency packages from the Humble distro.

**sudo apt install ros-foxy-hardware-interface
sudo apt install ros-foxy-xacro
sudo apt install ros-foxy-imu-sensor-broadcaster
sudo apt install ros-foxy-diff-drive-controller
sudo apt install ros-foxy-position-controllers
sudo apt install ros-foxy-gripper-controllers
sudo apt install ros-foxy-joint-state-broadcaster
sudo apt install ros-foxy-joint-trajectory-controller
sudo apt install ros-foxy-controller-manager**

and run **sudo apt-get update  && sudo apt-get dist-upgrade**.

ozanaki, 2023-07-17:
@yeongkyuyoon 
I have installed these packages for Humble as you suggested. This error message has gone, the manipulator has gone to the start position while brought up. 

**sudo apt install ros-humble-hardware-interface
sudo apt install ros-humble-xacro
sudo apt install ros-humble-imu-sensor-broadcaster
sudo apt install ros-humble-diff-drive-controller
sudo apt install ros-humble-position-controllers
sudo apt install ros-humble-gripper-controllers
sudo apt install ros-humble-joint-state-broadcaster
sudo apt install ros-humble-joint-trajectory-controller
sudo apt install ros-humble-controller-manager**

But this time some different error messages appeared on the terminal as 
`[ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!`
and
`[ros2_control_node-1] *** stack smashing detected ***: terminated`
 seems not related to this issue. I just proceed to another issue :) 

Thank you for your help

gandradeng, 2023-07-17:
I was able to solve the original issue but typing (at tb3) `sudo apt install ros-humble-launch-*`. But then another problem came through bring up:  
`[ros2_control_node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!`

yeongkyuyoon, 2023-07-18:
@ozanaki  @gandradeng 
Hi, I guess it's a problem with OpenCR's firmware. What firmware are you using? Have you installed it according to the guidelines in the E-manual?

> [TurtleBot3 SBC] Download the OpenCR firmware file on Raspberry Pi (SBC) and upload the correct firmware with the following commands.
> $ export OPENCR_PORT=/dev/ttyACM0
> $ export OPENCR_MODEL=turtlebot3_manipulation
> $ rm -rf ./opencr_update.tar.bz2
> $ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2
> $ tar -xvf opencr_update.tar.bz2
> $ cd ./opencr_update
> $ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr

gandradeng, 2023-07-18:
Hi, @yeongkyuyoon! I did exactly as in E-manual and until now the open manipulator is not working.

ozanaki, 2023-07-19:
@yeongkyuyoon 
I am exactly following the instructions in the manual including setting up the OpenCR. I installed the manipulation package on freshly installed ROS2 Humble on Ubuntu 22.04 on Raspberry Pi 4 (waffle pi with manipulator) described as in the manual  [https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#software-setup](https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#software-setup). 

Here is the ssh terminal output screen after applying the bringup command
`$ ros2 launch turtlebot3_manipulation_bringup hardware.launch.py`:
>[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-07-19-09-35-37-701805-wafflepi-1953
>[INFO] [launch]: Default logging verbosity is set to INFO
>[INFO] [ros2_control_node-1]: process started with pid [1956]
>[INFO] [robot_state_publisher-2]: process started with pid [1958]
>[INFO] [spawner-3]: process started with pid [1960]
>[INFO] [ld08_driver-4]: process started with pid [1962]
>[robot_state_publisher-2] [INFO] [1689748540.081126855] [robot_state_publisher]: got segment base_footprint
>[robot_state_publisher-2] [INFO] [1689748540.084080957] [robot_state_publisher]: got segment base_link
>[robot_state_publisher-2] [INFO] [1689748540.086333806] [robot_state_publisher]: got segment base_scan
>[robot_state_publisher-2] [INFO] [1689748540.088819302] [robot_state_publisher]: got segment camera_link
>[robot_state_publisher-2] [INFO] [1689748540.090887115] [robot_state_publisher]: got segment camera_rgb_frame
>[robot_state_publisher-2] [INFO] [1689748540.092974003] [robot_state_publisher]: got segment camera_rgb_optical_frame
>[robot_state_publisher-2] [INFO] [1689748540.093049058] [robot_state_publisher]: got segment caster_back_left_link
>[robot_state_publisher-2] [INFO] [1689748540.093081150] [robot_state_publisher]: got segment caster_back_right_link
>[robot_state_publisher-2] [INFO] [1689748540.093108428] [robot_state_publisher]: got segment end_effector_link
>[robot_state_publisher-2] [INFO] [1689748540.093135131] [robot_state_publisher]: got segment gripper_left_link
>[robot_state_publisher-2] [INFO] [1689748540.093160668] [robot_state_publisher]: got segment gripper_right_link
>[robot_state_publisher-2] [INFO] [1689748540.093186483] [robot_state_publisher]: got segment imu_link
>[robot_state_publisher-2] [INFO] [1689748540.093211742] [robot_state_publisher]: got segment link1
>[robot_state_publisher-2] [INFO] [1689748540.093237075] [robot_state_publisher]: got segment link2
>[robot_state_publisher-2] [INFO] [1689748540.093261408] [robot_state_publisher]: got segment link3
>[robot_state_publisher-2] [INFO] [1689748540.093287037] [robot_state_publisher]: got segment link4
>[robot_state_publisher-2] [INFO] [1689748540.093311648] [robot_state_publisher]: got segment link5
>[robot_state_publisher-2] [INFO] [1689748540.093335852] [robot_state_publisher]: got segment wheel_left_link
>[robot_state_publisher-2] [INFO] [1689748540.093360444] [robot_state_publisher]: got segment wheel_right_link
>[ros2_control_node-1] [INFO] [1689748540.103276592] [resource_manager]: Loading hardware 'TurtleBot3ManipulationSystem'
>[ros2_control_node-1] [INFO] [1689748540.107125038] [resource_manager]: Initialize hardware 'TurtleBot3ManipulationSystem'
>[ros2_control_node-1] [INFO] [1689748540.113484645] [turtlebot3_manipulation]: Succeeded to open port
>[ros2_control_node-1] [INFO] [1689748540.121159295] [turtlebot3_manipulation]: Succeeded to set baudrate
>[ros2_control_node-1] [INFO] [1689748540.150026914] [turtlebot3_manipulation]: OpenCR Model Number 20480
>[ros2_control_node-1] [INFO] [1689748540.178978772] [turtlebot3_manipulation]: Connected manipulator
>[ros2_control_node-1] [INFO] [1689748540.207992482] [turtlebot3_manipulation]: Connected wheels
>[ros2_control_node-1] [INFO] [1689748540.208380904] [resource_manager]: Successful initialization of hardware 'TurtleBot3ManipulationSystem'
>[ros2_control_node-1] [INFO] [1689748540.209037639] [resource_manager]: 'configure' hardware 'TurtleBot3ManipulationSystem'
>[ros2_control_node-1] [INFO] [1689748540.209114601] [resource_manager]: Successful 'configure' of hardware 'TurtleBot3ManipulationSystem'
>[ros2_control_node-1] [INFO] [1689748540.209157434] [resource_manager]: 'activate' hardware 'TurtleBot3ManipulationSystem'
>[ros2_control_node-1] [INFO] [1689748540.209181397] [turtlebot3_manipulation]: Ready for start
>[ros2_control_node-1] [INFO] [1689748540.236964525] [turtlebot3_manipulation]: Wait for IMU re-calibration
>[ros2_control_node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!
>[ld08_driver-4] /dev/ttyUSB0    CP2102 USB to UART Bridge Controller
>[ld08_driver-4] /dev/ttyACM0    OpenCR Virtual ComPort in FS Mode
>[ld08_driver-4] FOUND LDS-02
>[ld08_driver-4] LDS-02 started successfully
>[ros2_control_node-1] [INFO] [1689748543.271573655] [turtlebot3_manipulation]: Joints and wheels torque ON
>[ros2_control_node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!
>[ros2_control_node-1] [INFO] [1689748543.351609203] [turtlebot3_manipulation]: Set profile acceleration and velocity to joints
>[ros2_control_node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!
>[ros2_control_node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!
>[ros2_control_node-1] [INFO] [1689748543.422589409] [turtlebot3_manipulation]: Set profile acceleration and velocity to gripper
[... cut ...]
>[ros2_control_node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!
>[ros2_control_node-1] *** stack smashing detected ***: terminated
>[ERROR] [ros2_control_node-1]: process has died [pid 1956, exit code -11, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_5fpdjqzo --params-file /home/ubuntu/turtlebot3_ws/install/turtlebot3_manipulation_bringup/share/turtlebot3_manipulation_bringup/config/hardware_controller_manager.yaml -r ~/cmd_vel_unstamped:=cmd_vel -r ~/odom:=odom'].

JanG175, 2023-07-31:
Hi, I've got the same problem as @ozanaki. Did anybody manage to solve it? I suppose it is something connected with OpenCR software.

AugensteinT, 2024-01-16:
Hi, I've got the same problem as @ozanaki. Did anybody manage to solve it?

krizchong, 2024-06-07:
I am also getting the same problem as @ozanaki , fail to bring up TB3 with Openmanipuator-X in ROS 2 Humble, getting error “[ros2_control_node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!”...

Any solution?

sunghowoo, 2024-09-17:
This issue occurs with the latest version of Dynamixel, and it has been confirmed that it can be resolved by using the Dynamixel Wizard to change the **Return Delay Time** of all motors from 250 ms to 0 ms.

closed 2024-09-17 by sunghowoo