ROBOTIS-GIT/turtlebot3_manipulation #69: Newer version of open manipulator arm – urdf and meshes

lmendyk, 2023-08-22:
Is there a chance to have an updated version of the model in respect to meshes and especially the geometry. 
I have a new version which robot arm looks like this https://emanual.robotis.com/assets/images/platform/openmanipulator_x/OpenManipulator_Introduction.jpg
But the model look more like this https://emanual.robotis.com/assets/images/platform/turtlebot3/manipulation/hardware_setup.png

The problem is not only that the turtlebot3_manipulation looks in rviz differently than in reality but I also suspect that this may be the reason why trajectories planned and controlled by moveIt when executed experience collisions with the base of the turtlebot which are not visible/detected by the planning stack.  I suspect that it may be caused by differences in geometry which are not huge but enough to have collisions not comprehend by the moveIt stack.

sunghowoo, 2024-09-16:
@lmendyk Apologies, but we currently do not manage an updated version of the mesh. While the new robot arm does have relatively rounded joints, if you could provide more details on which specific movements are causing collisions with the TurtleBot's base, we would be happy to investigate and debug the issue further.

sunghowoo, 2025-01-16:
This issue will be closed due to prolonged inactivity.

closed 2025-01-16 by sunghowoo