ROBOTIS-GIT/turtlebot3_manipulation #70: [Simulation, Humble] abrupt arm movement due to `open_loop_control` in `joint_trajectory_controller`

goekce, 2023-09-05:
I use 
```
ros2 launch turtlebot3_manipulation_bringup gazebo.launch.py
```
and control the arm directly using the `/arm_controller/joint_trajectory` topic. When I send the first command then it leads to an abrupt start as follows:

[image]

This behavior is probably due to:
https://github.com/ROBOTIS-GIT/turtlebot3_manipulation/blob/3baa265d0f4843955ed77716be6f88b71689d7a1/turtlebot3_manipulation_bringup/config/gazebo_controller_manager.yaml#L45

[ROS2 Control docs](https://control.ros.org/master/doc/ros2_controllers/joint_trajectory_controller/doc/parameters.html) state:

> If this flag is set, the controller tries to read the values from the command interfaces on activation. If they have real numeric values, those will be used instead of state interfaces. Therefore it is important set command interfaces to NaN (i.e., std::numeric_limits<double>::quiet_NaN()) or state values when the hardware is started.

Setting `open_loop_control` to `false` fixes the issue.

Why is `open_loop_control` is active? Is a feedback from the hardware interface not better?

goekce, 2024-01-31:
@ROBOTIS-Will will Turtlebot3 receive any updates in the future for ROS2?

Background: Currently most of the effort seems to be driven by the community, however Turtlebot3 has a complex infrastructure, which is difficult to understand/contribute. If Turtlebot3 won't receive any focus from your dev team in the near future, please state this in your README, or pin an issue.

sunghowoo, 2024-09-17:
@goekce sorry for late response. The advantage of using open loop control is that the system can execute commands immediately, without waiting for feedback from the hardware. This allows for quicker execution, particularly in situations where real-time feedback isn't critical. However, if you set the initial position of the /arm_controller/joint_trajectory correctly, abrupt movements can be avoided, even when using open loop control. As long as the initial command matches the actual state of the hardware, smooth movements will be ensured. Alternatively, you can opt for closed loop control, which uses real-time feedback from the hardware to provide more stable and smooth motion. Depending on the requirements, you can choose the most appropriate method.

Our team is planning to resume maintaining the open platform products, including TurtleBot3. We apologize for not being able to provide service in this area for some time. We will soon update the README, as well as the manuals and source code for the latest versions.

goekce, 2024-09-17:
Great that Robotis will have more resources for Turtlebots, even I don't have any lectures using Turtlebots anymore 😐.

@sunghowoo would be great if you could drop a short snippet how the initial position of the `/arm_controller/joint_trajectory` can be set correctly.

closed 2024-09-17 by goekce