ROBOTIS-GIT/turtlebot3_manipulation #71: FAIL to start "moveit_servo" (ROS2 Foxy)

eRobotclub, 2024-01-29:
I am trying with ROS2 Foxy on TB3 with OpenManipulator-X, and followed the instructions in emanual for all setup.
However, when I try to teleoperate the arm, but error: FAIL to start FAIL to start "moveit_servo"

T1: ros2 launch turtlebot3_manipulation_bringup hardware.launch.py   ==> No error
T2: ros2 launch turtlebot3_manipulation_moveit_config servo.launch.py ==> No error
T3: ros2 run turtlebot3_manipulation_teleop turtlebot3_manipulation_teleop ==> ERROR!!!
T3: ros2 service list ==> Can get /servo_server...

[image]

Why this ERROR happen? Anything I missed out?

sunghowoo, 2024-06-02:
It seems to me that the issue is that it can't find the moveit_servo package. Try reinstalling the moveit2 package from source from [moveit2](https://github.com/moveit/moveit2?tab=readme-ov-file) and give it another shot.

eRobotclub, 2024-06-11:
> It seems to me that the issue is that it can't find the moveit_servo package. Try reinstalling the moveit2 package from source from [moveit2](https://github.com/moveit/moveit2?tab=readme-ov-file) and give it another shot.

I cannot regenerate this problem anymore because after I reinstall the micro SD card with either ROS2 Foxy or ROS2 Humble, I cannot bring up the TB3 with OpenManipulator-X at all! It keeps showing "[ros2_control_node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!", similar like the issue in https://github.com/ROBOTIS-GIT/turtlebot3_manipulation/issues/63! But there is no update on that issue!!!

sunghowoo, 2024-06-12:
@eRobotclub  I'm curious if Dynamixel and OpenCR are communicating properly. When you turn on your OpenCR, do the manipulator and both wheels remain stationary due to torque when touched?

eRobotclub, 2024-06-12:
> @eRobotclub I'm curious if Dynamixel and OpenCR are communicating properly. When you turn on your OpenCR, do the manipulator and both wheels remain stationary due to torque when touched?

Yes, when power on the OpenCR, all the Dynamixel on wheels and manipulator are TORQUE ON.

And if I upload the OpenCR firmware for waffle pi (without manipulation) to the robot, I can bringup and teleoperate TB3 with no problem, therefore the USB connection between Raspberry Pi and OpenCR is OK.
And if I use Androind IDE to search connected Dynamixels on OpenCR, I can found all ID 1, 2, 11, 12, 13, 14, 15 motors, therefore the connection between Dynamixels and OpeCR also no problem.

Furthermore, I am not the only one facing the same problem: https://github.com/ROBOTIS-GIT/turtlebot3_manipulation/issues/63
so maybe some issue after ROS2 updated? Or issue on OpenCR firmware for ROS2? I have no idea how to fix this...

sunghowoo, 2024-06-12:
@eRobotclub I understand your difficulties. I tried to reproduce this issue using ROS2 Humble version, but I was unable to replicate it. First, I reinstalled Humble on both Turtlebot3 and the remote PC from scratch following the e-manual. When I executed the command "ros2 launch turtlebot3_manipulation_bringup hardware.launch.py" on the Turtlebot, the "[DynamixelSDKWrapper] [TxRxResult] There is no status packet!" log appeared only once initially and did not persist.

The ROS2 firmware for OpenCR (OPENCR_MODEL=turtlebot3_manipulation) seems to be working correctly. Try deleting the opencr_update folder in your home directory and update it again. Here are my logs:

Regarding the issue you mentioned initially, it can occur if your Turtlebot and remote PC are not properly connected. Ensure that the ROS_DOMAIN_ID is the same on both devices. Additionally, 5G WiFi may not work well for this connection, so I recommend using a hotspot to connect them.

> turtlebot@robotis:~$ ros2 launch turtlebot3_manipulation_bringup hardware.launch.py
[INFO] [launch]: All log files can be found below /home/turtlebot/.ros/log/2024-06-12-18-59-51-966397-robotis-2872
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [2875]
[INFO] [robot_state_publisher-2]: process started with pid [2877]
[INFO] [spawner-3]: process started with pid [2879]
[INFO] [ld08_driver-4]: process started with pid [2881]
[robot_state_publisher-2] Link base_link had 8 children
[robot_state_publisher-2] Link link1 had 1 children
[robot_state_publisher-2] Link link2 had 1 children
[robot_state_publisher-2] Link link3 had 1 children
[robot_state_publisher-2] Link link4 had 1 children
[robot_state_publisher-2] Link link5 had 3 children
[robot_state_publisher-2] Link end_effector_link had 0 children
[robot_state_publisher-2] Link gripper_left_link had 0 children
[robot_state_publisher-2] Link gripper_right_link had 0 children
[robot_state_publisher-2] Link camera_link had 1 children
[robot_state_publisher-2] Link camera_rgb_frame had 1 children
[robot_state_publisher-2] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher-2] Link caster_back_left_link had 0 children
[robot_state_publisher-2] Link caster_back_right_link had 0 children
[robot_state_publisher-2] Link imu_link had 0 children
[robot_state_publisher-2] Link base_scan had 0 children
[robot_state_publisher-2] Link wheel_left_link had 0 children
[robot_state_publisher-2] Link wheel_right_link had 0 children
[robot_state_publisher-2] [INFO] [1718186395.156581212] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-2] [INFO] [1718186395.157400842] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1718186395.157539045] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-2] [INFO] [1718186395.157578675] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-2] [INFO] [1718186395.157611045] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-2] [INFO] [1718186395.157642249] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher-2] [INFO] [1718186395.157672490] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher-2] [INFO] [1718186395.157705231] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher-2] [INFO] [1718186395.157734101] [robot_state_publisher]: got segment end_effector_link
[robot_state_publisher-2] [INFO] [1718186395.157762768] [robot_state_publisher]: got segment gripper_left_link
[robot_state_publisher-2] [INFO] [1718186395.157790508] [robot_state_publisher]: got segment gripper_right_link
[robot_state_publisher-2] [INFO] [1718186395.157819286] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-2] [INFO] [1718186395.157846749] [robot_state_publisher]: got segment link1
[robot_state_publisher-2] [INFO] [1718186395.157878490] [robot_state_publisher]: got segment link2
[robot_state_publisher-2] [INFO] [1718186395.158123268] [robot_state_publisher]: got segment link3
[robot_state_publisher-2] [INFO] [1718186395.158234064] [robot_state_publisher]: got segment link4
[robot_state_publisher-2] [INFO] [1718186395.158272342] [robot_state_publisher]: got segment link5
[robot_state_publisher-2] [INFO] [1718186395.158303693] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-2] [INFO] [1718186395.158339212] [robot_state_publisher]: got segment wheel_right_link
[ros2_control_node-1] [INFO] [1718186395.231406582] [resource_manager]: Loading hardware 'TurtleBot3ManipulationSystem' 
[ros2_control_node-1] [INFO] [1718186395.259100564] [resource_manager]: Initialize hardware 'TurtleBot3ManipulationSystem' 
[ros2_control_node-1] [INFO] [1718186395.264583175] [turtlebot3_manipulation]: Succeeded to open port
[ros2_control_node-1] [INFO] [1718186395.272265916] [turtlebot3_manipulation]: Succeeded to set baudrate
[ros2_control_node-1] [INFO] [1718186395.287146323] [turtlebot3_manipulation]: OpenCR Model Number 20480
[ros2_control_node-1] [INFO] [1718186395.288756342] [turtlebot3_manipulation]: Connected manipulator
[ros2_control_node-1] [INFO] [1718186395.292072842] [turtlebot3_manipulation]: Connected wheels
[ros2_control_node-1] [INFO] [1718186395.292479175] [resource_manager]: Successful initialization of hardware 'TurtleBot3ManipulationSystem'
[... cut ...]
[spawner-7] [INFO] [1718186401.961028597] [spawner_imu_broadcaster]: Loaded imu_broadcaster
[ros2_control_node-1] [INFO] [1718186401.971154189] [controller_manager]: Configuring controller 'imu_broadcaster'
[INFO] [spawner-8]: process has finished cleanly [pid 2930]
[spawner-7] [INFO] [1718186402.086606837] [spawner_imu_broadcaster]: Configured and activated imu_broadcaster
[INFO] [spawner-5]: process has finished cleanly [pid 2924]
[INFO] [spawner-6]: process has finished cleanly [pid 2926]
[INFO] [spawner-7]: process has finished cleanly [pid 2928]

sunghowoo, 2024-06-12:
Also, when you updated the firmware, did it update successfully like the log below?
[image]

eRobotclub, 2024-06-18:
@sunghowoo 
I burn again the Ubuntu Server 22.04 to the micro SD card again~
Then setup and install ROS2 Humble with following commands:

sudo nano /etc/netplan/50-cloud-init.yaml
sudo nano /etc/apt/apt.conf.d/20auto-upgrades
systemctl mask systemd-networkd-wait-online.service
sudo systemctl mask sleep.target suspend.target hibernate.target hybrid-sleep.target
reboot
locale
sudo apt update && sudo apt install locales
sudo locale-gen en_US en_US.UTF-8
sudo update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
export LANG=en_US.UTF-8
locale
sudo apt install software-properties-common
sudo add-apt-repository universe
sudo apt update && sudo apt install curl -y
sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UB>
sudo apt update
sudo apt upgrade -y
sudo apt install ros-humble-desktop
sudo apt install ros-humble-ros-base
sudo apt install ros-dev-tools
source /opt/ros/humble/setup.bash
sudo apt install python3-argcomplete python3-colcon-common-extensions libboost-system-dev build-essential
sudo apt install ros-humble-hls-lfcd-lds-driver
sudo apt install ros-humble-turtlebot3-msgs
sudo apt install ros-humble-dynamixel-sdk
sudo apt install libudev-dev
sudo reboot
mkdir -p ~/turtlebot3_ws/src && cd ~/turtlebot3_ws/src
git clone -b humble-devel https://github.com/ROBOTIS-GIT/turtlebot3.git
git clone -b ros2-devel https://github.com/ROBOTIS-GIT/ld08_driver.git
git clone -b humble-devel https://github.com/ROBOTIS-GIT/turtlebot3_manipulation.git
cd ~/turtlebot3_ws/src/turtlebot3
rm -r turtlebot3_cartographer turtlebot3_navigation2
echo 'source /opt/ros/humble/setup.bash' >> ~/.bashrc
source ~/.bashrc
cd ~/turtlebot3_ws/
sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src -r -y
source ~/.bashrc
colcon build --symlink-install --parallel-workers 1
echo 'source ~/turtlebot3_ws/install/setup.bash' >> ~/.bashrc
source ~/.bashrc
sudo udevadm control --reload-rules
sudo udevadm trigger
source ~/.bashrc
echo 'export LDS_MODEL=LDS-01' >> ~/.bashrc
source ~/.bashrc
sudo dpkg --add-architecture armhf
sudo apt update
sudo apt install libc6:armhf
cd ~
export OPENCR_MODEL=turtlebot3_manipulation
rm -rf ./opencr_update.tar.bz2
wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2
tar -xvf opencr_update.tar.bz2
cd ./opencr_update
./update.sh $OPENCR_PORT $OPENCR_MODEL.opencrcd ./opencr_update
cd ~
source ~/.bashrc
ros2 launch turtlebot3_manipulation_bringup hardware.launch.py

I used rosdep to install some package because if just following the instruction in emanual, some instruction such as hardware-interface is not installed!

Then I get the error when bring up:
[... 94 lines cut ...]
[INFO] [spawner-8]: process started with pid [9092]
[spawner-7] [INFO] [1718676055.083555305] [spawner_imu_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1718676055.146858214] [spawner_diff_drive_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1718676055.164711388] [spawner_gripper_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1718676055.235554405] [spawner_arm_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1718676057.145886860] [spawner_imu_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1718676057.207541042] [spawner_diff_drive_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1718676057.209342546] [spawner_gripper_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1718676057.295828958] [spawner_arm_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1718676059.206192213] [spawner_imu_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1718676059.266002261] [spawner_gripper_controller]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1718676059.266002575] [spawner_diff_drive_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1718676059.355159022] [spawner_arm_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1718676061.265914796] [spawner_imu_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1718676061.327309812] [spawner_diff_drive_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1718676061.327309275] [spawner_gripper_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1718676061.414674551] [spawner_arm_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [ERROR] [1718676063.115611313] [spawner_imu_broadcaster]: Controller manager not available
[spawner-5] [ERROR] [1718676063.179945936] [spawner_gripper_controller]: Controller manager not available
[spawner-8] [ERROR] [1718676063.179945899] [spawner_diff_drive_controller]: Controller manager not available
[spawner-6] [ERROR] [1718676063.268298574] [spawner_arm_controller]: Controller manager not available
[ERROR] [spawner-7]: process has died [pid 9090, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner imu_broadcaster --ros-args'].
[ERROR] [spawner-5]: process has died [pid 9086, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner gripper_controller --ros-args'].
[ERROR] [spawner-8]: process has died [pid 9092, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner diff_drive_controller -c /controller_manager --ros-args'].
[ERROR] [spawner-6]: process has died [pid 9088, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner arm_controller --ros-args'].

eRobotclub, 2024-06-18:
I don't know why it show "[ros2_control_node-1] [FATAL] [1718676039.739130871] [turtlebot3_manipulation]: Not connected manipulator"?

I use Arduino IDE to scan on OpenCR, I get all 7 Dynamixels...
[image]

sunghowoo, 2024-09-16:
This issue occurs with the latest version of Dynamixel, and it has been confirmed that it can be resolved by using the Dynamixel Wizard to change the **Return Delay Time** of all motors from 250 ms to 0 ms.

closed 2024-09-16 by sunghowoo