ROBOTIS-GIT/turtlebot3_manipulation #76: Turtlebot Waffle Pi from Openmanipulator with Gazebo

Alleekkss, 2025-01-18:
Hello, currently working in ROS Noetic
I have a problem with lifting a can and moving it to another place in Gazebo
The robot model is Turtlebot Waffle Pi from Openmanipulator
I want to do something similar to this repository: https://github.com/AuTURBO/open_manipulator_with_tb3
I already have a script with reaching the right point, all I need is to lift and move the can and if you could I would like to make it so that you detect the Aruco marker using the camera before lifting it
Maybe someone can help with this?

[image]

Script with getting to the point:

`#!/usr/bin/env python3
 
import rospy
from geometry_msgs.msg import PoseStamped
 
def send_goal(x, y, theta):
    # Publisher dla topicu /move_base_simple/goal
    goal_pub = rospy.Publisher('/move_base_simple/goal', PoseStamped, queue_size=10)
    # Oczekiwanie na połączenie z `move_base`
    rospy.sleep(2)  
 
    # Tworzenie celu
    goal = PoseStamped()
    goal.header.frame_id = "map"
    goal.header.stamp = rospy.Time.now()
    goal.pose.position.x = x
    goal.pose.position.y = y
    goal.pose.orientation.z = theta  # Uwaga: pełny kwaternion może być potrzebny
 
    # Publikacja celu
    rospy.loginfo(f"Wysyłanie celu: x={x}, y={y}, theta={theta}")
    goal_pub.publish(goal)
 
def main():
    rospy.init_node('multi_point_navigation', anonymous=True)
 
    while not rospy.is_shutdown():
        try:
            # Pobieranie współrzędnych od użytkownika
            x = float(input("Podaj współrzędną x: "))
            y = float(input("Podaj współrzędną y: "))
            theta = float(input("Podaj orientację (theta): "))
           
            # Wysyłanie celu
            send_goal(x, y, theta)
 
            # Pytanie o kontynuację
            cont = input("Czy chcesz wprowadzić kolejny punkt? (t/n): ").strip().lower()
            if cont != 't':
                rospy.loginfo("Koniec wprowadzania punktów.")
                break
        except ValueError:
            rospy.logwarn("Niepoprawny format danych. Spróbuj ponownie.")
        except rospy.ROSInterruptException:
            rospy.loginfo("Przerwano działanie skryptu.")
            break
 
if __name__ == "__main__":
    main()
`

sunghowoo, 2025-01-21:
@Alleekkss This manual may still be helpful, although it is no longer maintained. Wishing you the best of luck!
👉 [TurtleBot3 Home Service Challenge Manual](https://emanual.robotis.com/docs/en/platform/turtlebot3/home_service_challenge/#turtlebot3-home-service-challenge)

Alleekkss, 2025-01-21:
@sunghowoo I tried but nothing worked

sunghowoo, 2025-02-24:
@Alleekkss Apologies for the delayed response. Currently, the Home Service Challenge is no longer maintained in ROS 1 Noetic. Our Open Platform Team, which was newly established in January, has been working on porting it to ROS 2 Humble. This porting process is expected to be completed by April.

sunghowoo, 2025-04-15:
@Alleekkss
Since the Humble version of the Home Service Challenge has been merged, I will go ahead and close this issue.
 https://github.com/ROBOTIS-GIT/turtlebot3_home_service_challenge/pull/29

closed 2025-04-15 by sunghowoo