ROBOTIS-GIT/turtlebot3_manipulation #82: turtlebot3_manipulation model not showing in gazebo

sento2310, 2025-04-11:
Hello,
I started working with the turtlebot3 waffle_pi and want to use the model with manipulation in gazebo.
As setup I am running ubuntu 22.04.05 LTS in virtual box.
I followed the installation guides on the emanual.robotis.com website, installing the ROS 2 debian package and the dependent ros2 packages.
Then I installed the turtlebot3 packages.
After this, turtlebot3_manipulation was installed too.

Now when running the turtlebot3_manipulation_bringup with the gazebo_launch.py, Gazebo starts up, the world loads and I can see the blue lidar rays. But at the position where the turtlebot is supposed to be, I see a transparent circle, as seen in the picture below
 
[image]

When running the command  with the addition start_rviz:=true, the same issue persists in gazebo. In RViz I can see the the turtlebot with manipulator.

I set up a new virtualbox and repeated the installation process several times, but this issue is present each time.
Is this problem familiar? Are there any suggestions on how to solve this?
I would appreciate your help

sunghowoo, 2025-04-15:
It looks like Gazebo can’t properly load the TurtleBot3 Waffle Pi model, which is why you’re only seeing a transparent circle. This usually happens when the TURTLEBOT3_MODEL environment variable isn’t set (e.g., export TURTLEBOT3_MODEL=waffle_pi) or the mesh files are missing or not correctly sourced.

Running Gazebo with TurtleBot3 manipulation in a VirtualBox environment is generally not recommended due to limited graphics acceleration and potential issues with 3D rendering.

sento2310, 2025-04-15:
Thank you for your answer. 
I have export TURTLEBOT3_MODEL=waffle_pi in the bashrc all the time.

I tried it on a native linux system now.
There I get exactly the same issue. 
I can make the turtlebot with manipulator appear in gazebo, if I change the URDF files of turtlebot3_manipulation.
As soon as I added following property to them, it spawned into Gazebo.

` 
  <xacro:property
    name="meshes_file_direction"
    value="package://turtlebot3_manipulation_description/meshes/turtlebot3_waffle_pi" />`

But when I start the simulation with Moveit, the manipulator won´t load into the rviz window.
Also cartographer will still have the TF errors with the missing transforms.

GyuH13, 2025-04-16:
@sento2310 Hello
Can I get the terminal status image and the TF Tree captured through rqt after launching Gazebo?

sento2310, 2025-04-16:
@GyuH13 Hello,
Here is the tf tree:

[image]

The Gazebo logs are in this pastebin: [https://pastebin.com/JFVSMxuu](url)

Sheero-git, 2025-04-30:
I have the same issue
When I start gazebo I get these errors
Also cannot use Move it
Any idea what to do?

[gzserver-3] [INFO] [1746030835.229042777] [camera_controller]: Publishing camera info to [/pi_camera/camera_info]
[spawn_entity.py-5] [INFO] [1746030835.324295417] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [turtlebot3_manipulation_system]
[INFO] [spawn_entity.py-5]: process has finished cleanly [pid 42581]
[spawner-2] [WARN] [1746030843.655140102] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
[spawner-2] [INFO] [1746030843.655475801] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[spawner-2] [WARN] [1746030853.675103012] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
[spawner-2] [INFO] [1746030853.675431611] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[spawner-2] [WARN] [1746030863.692297477] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
[spawner-2] [INFO] [1746030863.692814475] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[spawner-2] [WARN] [1746030873.710805301] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
[spawner-2] [INFO] [1746030873.711158800] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...

naroasc, 2025-05-27:
Hi,

I’m experiencing the same issue ((Ubuntu 22.04, Ros2 Humble)

Yesterday it was working without any problems, and today (without having changed anything) it suddenly doesn’t work anymore.

I had saved a previous version of the package that still works perfectly. The problem only appears when I try to use the latest version.

It seems the issue might be related to how Gazebo is launched: if it opens quickly, everything works fine; but if it takes longer to load, the model doesn’t load properly and these issues occur.

Any ideas on how to fix this?

Sheero-git, 2025-05-27:
Yes, the issue persists

I have discovered a workaround which isn't the best but works fine for my purposes
I only use the manipulation package from another branch which is humble-devel

Here are the bash commands to do it:

cd ~/turtlebot3_ws/src/turtlebot3_manipulation (your path to manipulation package)
git stash
mkdir -p ~/tmp_branch_swap
git checkout humble-devel
cp -r turtlebot3_manipulation_description ~/tmp_branch_swap/
git checkout humble
rm -rf turtlebot3_manipulation_description
cp -r ~/tmp_branch_swap/turtlebot3_manipulation_description .
cd ~/turtlebot3_ws
colcon build --packages-select turtlebot3_manipulation_description
source install/setup.bash

Let me know if it corrects the error for you.
Best of Luck!

Sheero-git, 2025-05-28:
I haven’t specifically tried this but I guess you can use teleop launch file to try commanding the turtlebot. If it successfully moves the bot without any curvature then we isolate the problem to the way you’re publishing to /cmd_vel
Also, if teleop is successful, echo /cmd_vel to understand the correct sequence of commands to command the vehicle. 
Hope that helps!

naroasc, 2025-05-28:
Thank you for your help!

I have tried teleop and it still makes the curvature. I think it may be something related to friction because if the velocity is small, the curvature is reduced. However, I have tried to use the same gazebo characteristics of the package turtlebot (without the manipulator), and it still makes the curvature, which is why I find it strange. If I modified mu1 and mu2, I can reduce the error, but it is still there. 

Maybe there is something more in comparison with the turtlebot package, but I can't find it.

GyuH13, 2025-07-11:
Hello @sento2310 , @Sheero-git , @naroasc 

We have identified and resolved the issue.

The problem occurs because spawn_entity.py from gazebo_ros replaces the package:// syntax in the robot description with model:// when loading the URDF. Upon reviewing the spawn_entity.py code, we found that it works only when a specific parameter is used—though the reason why it works in this way is still unclear.

To address this issue, we added an export of the GAZEBO_MODEL_PATH.

Additionally, please note that the simulation-related packages for tb3_manipulation have been fully migrated to the tb3_simulation repository. Kindly refer to the manual and use the tb3_simulation repository going forward.

https://github.com/ROBOTIS-GIT/turtlebot3_simulations/pull/246

Thank you! And this issue is now closed.

closed 2025-07-11 by GyuH13